Knio/pynmea2 #54: Decimal lat and lon ?

dbrb2, 2016-06-11:
Hi,

I have installed this library via pip, but have a problem accessing decimal lat and lon....

The below: 
msg = pynmea2.parse(rcv)

print msg.gps_qual
print rcv
print msg.lat+" "+msg.lon
print msg.latitude+" "+msg.longitude

Gives the following output: 
1
$GPGGA,185953.00,5123.94863,N,00009.08705,W,1,08,1.18,47.3,M,45.6,M,,*7E
5131.94862 00008.08693

In other words the un-converted lat/lon is printed, but not the decimal version. Any idea why...?

Knio, 2016-06-11:
Hi,

This sentence seems to work fine for me. I don't see your last line of output there?

```
>>> r = pynmea2.parse('$GPGGA,185953.00,5123.94863,N,00009.08705,W,1,08,1.18,47.3,M,45.6,M,,')
>>> r.latitude
51.39914383333333
>>> r.longitude
-0.15145083333333334
>>> r.lat
'5123.94863'
>>> r.lon
'00009.08705'
```

dbrb2, 2016-06-11:
Ah - sorted - it was a case of "error in front of keyboard"

I was using:

print msg.latitude+" "+msg.longitude

To get my output - without explicitly casting the lat / lon to strings
beforehand. For some reason this didn't throw an error, but would explain
why I was getting no output!

All OK now - apologies, but thanks for trying to help so quickly!

closed 2016-06-12 by Knio