Knio/pynmea2 #55: Fails on noisy input

troxel, 2016-11-02:
I get a failure trying to access a property on a bad input... Looks like checksum is checked only if it exists in string. If no checksum is in the string the package will still try to parse but fail when trying to access properties. For example attached file will fail when trying to access the properties. 

[test_bad_rmc.py.txt](https://github.com/Knio/pynmea2/files/565613/test_bad_rmc.py.txt)

Version 1.5.3

closed 2016-11-02 by Knio

Knio, 2016-11-02:
It's unclear what the expected behavior is here when you're passing it corrupt input.

I added a strict checking argument to `parse()` that will enforce that the data has a valid checksum, if you want to fast-fail, and I also suppressed the exception coming from the automatic type conversion so that it is consistent and will just return the corrupt data field, similar to how string (or missing) fields behave already.

Let me know if this resolves your issue!

troxel, 2016-11-02:
I am parsing very noise data over a radio link. I fixed my problem by just
adding an else: after the if checksum: and raising a checksum error. That
is if no checksum is returned from the match raise error. Does that make
sense?

Knio, 2016-11-02:
Yes makes sense. that sounds the same as the new `parse(data, check=True)` flag I added here, so you can use that if you upgrade to 1.6.0 now.

commit 6c8df584 (2016-11-02): Suppress automatic type conversions if they fail on corrupt data. Add strict parsing flag, which fails on messages that don't have a checksum. Fixes #55