Knio/pynmea2 #56: Unknown sentence type GPPWR

stuporglue, 2016-11-04:
I'm using a Dual XGPS160 and getting the following issue 

```
Traceback (most recent call last):
  File "./test.py", line 14, in <module>
    msg = pynmea2.parse( data )
  File "/Library/Python/2.7/site-packages/pynmea2/nmea.py", line 130, in parse
    raise SentenceTypeError('Unknown sentence type %s' % sentence_type)
pynmea2.nmea.SentenceTypeError: Unknown sentence type GPPWR,
```

The sentence is: 

```
$GPPWR,0922,1,0,0,0,00,0,S,97, 1 5 ,000*1F
```

Thanks,
Michael

jmwooten, 2016-11-04:
The 'PWR' sentence type is new to me. Do you have any field information?

stuporglue, 2016-11-04:
Sorry, 

I'm new to parsing NMEA and am still getting up to speed. 

It looks like PWR is a proprietary addition to the XGPS160 (and other models from Dual? Other brands? No idea.). 

Their docs say: 

> The PPWR sentence contains device specific information and looks like this:
> $GPPWR,0876,0,0,0,0,00,F,0,97,1,3,000,00190108EEEE,0017E9B92122*74
> • Element #1, 0876, is the battery voltage. Battery voltage is not valid while the device is charging.
> • Element #5 is the charging status: 1 = charging, 0 = not charging.

That comes from the SDK documentation which isn't on their site, we had to request it.

kamiccolo, 2017-10-11:
Yet, another proprietary extension $PPWR:

> $PPWR,vv.vv,i,tt*hh<CR><LF>
> vv.vv is the modem supply voltage in volts.
> i is the ignition status: 0 = off; 1 = on.
> tt is the modem’s internal temperature in degrees Celsius.
> hh is the message checksum

> Example:
> $PPWR,14.12,1,43*37

http://www.cypress.bc.ca/documents/Report_Messages/CTM200/msg_116_PPWR.html