Knio/pynmea2 #63: streamreader problem

jeremy-rutman, 2017-04-20:
I am getting lines like the following

```
$GPGGA,183800.0,3203.121,N,03446.075,E,1,03,4.90,00038,M,017,M,,*5A
$GPVTG,000.0,T,357.3,M,000.00,N,000.00,K*4C
```
from serial read of a trimble gps . I tried 
```
streamreader = pynmea2.NMEAStreamReader()
while 1:
        line = ser.readline().decode('ascii', errors='replace')
#       print (line.strip())
        print (line)
        if line[0:6]=='$GPGGA':
                msg = pynmea2.parse(line.strip())
                print('msg1:'+str(msg))
        for msg in streamreader.next(line.strip()):
                print('msg:'+str(msg))
                print('lat {} long {}'.format(msg.latitude,msg.longitude))
```
The pynmea2.parse spits back the input, while the streamreader.next does not get any msgs. Any clues?

jmwooten, 2017-04-20:
streamreader.next() expects new lines at the end of each NMEASentence.

Try using read() instead of readline().

Example with python 2.7
```python
#!/usr/bin/env python
import pynmea2
import StringIO
lines="$GPGGA,183800.0,3203.121,N,03446.075,E,1,03,4.90,00038,M,017,M,,*5A\n $GPVTG,000.0,T,357.3,M,000.00,N,000.00,K*4C\n"

buf = StringIO.StringIO(lines)

print "Passing Lines from readlines()"
streamreader = pynmea2.NMEAStreamReader()
for line in buf.readlines():
    if line[0:6]=='$GPGGA':
            msg = pynmea2.parse(line.strip())
            print('msg1:'+str(msg))
    for msg in streamreader.next(line.strip()):
            print('msg:'+str(msg))

buf = StringIO.StringIO(lines)
print "Passing Lines from read()"
streamreader = pynmea2.NMEAStreamReader()
for msg in streamreader.next(buf.read()):
        print('msg:'+str(msg))
```

Yields
```
 Passing Lines from readlines()
 msg1:$GPGGA,183800.0,3203.121,N,03446.075,E,1,03,4.90,00038,M,017,M,,*5A
 Passing Lines from read()
 msg:$GPGGA,183800.0,3203.121,N,03446.075,E,1,03,4.90,00038,M,017,M,,*5A
 msg:$GPVTG,000.0,T,357.3,M,000.00,N,000.00,K*4C
```

Knio, 2017-04-21:
You should either use `streamreader` **or** `parse()`, but not both. Trying to use both at the same time makes no sense. If you want to do your own IO, then just call `parse()` for each line. 

Checking for `$GPGGA` is also redundant, as the whole point of this library is to parse the message for you. you can parse each message and check `msg.sentence_type == 'GGA'` or `isinstance(msg, pynmea2.GGA)`

closed 2017-04-21 by Knio