Knio/pynmea2 #79: I can not import pynmea2

galev00, 2018-11-28:
Hello,
I'm new to Python and
I couldn't import "pynmea". What can i do?

I'm using Raspberry Pi 3 B+ with noobs 3

**for install i used**
pi@raspberrypi:~ $ pip install pynmea2
Collecting pynmea2
  Downloading https://files.pythonhosted.org/packages/7c/82/f734e75dbb0aecbd9ea20489f364fdac9f124f17c4d4e26d2ab160c8ff58/pynmea2-1.12.0.tar.gz
Building wheels for collected packages: pynmea2
  Running setup.py bdist_wheel for pynmea2 ... done
  Stored in directory: /home/pi/.cache/pip/wheels/4f/3f/c4/4748d48cbe607279db39640f9295a9eaf12f37c21716272066
Successfully built pynmea2
Installing collected packages: pynmea2
Successfully installed pynmea2-1.12.0
pi@raspberrypi:~ $ 

**in python i used this**
# pynmea to read
import pynmea

**when i run it**
Python 3.5.3 (/usr/bin/python3)
>>> %Run screen7.py
Traceback (most recent call last):
  File "/home/pi/Desktop/screen7.py", line 3, in <module>
    import pynmea
  File "/usr/lib/python3/dist-packages/thonny/backend.py", line 317, in _custom_import
    module = self._original_import(*args, **kw)
ImportError: No module named 'pynmea'
>>>

xOneca, 2018-11-28:
You have to `import pynmea2` (note the number 2 at the end).

galev00, 2018-11-28:
Really thank you for your answer but that also didn't work

Traceback (most recent call last):
  File "/home/pi/Desktop/screen5-4.py", line 3, in <module>
    import pynmea2
  File "/usr/lib/python3/dist-packages/thonny/backend.py", line 317, in _custom_import
    module = self._original_import(*args, **kw)
ImportError: No module named 'pynmea2'
>>>

xOneca, 2018-11-28:
Maybe installing the package in the system path? (running `pip` as root) I suspect it is installed in `~/.local` and this path is not in Python's `sys.path`.

Knio, 2018-12-23:
As others noted, this is an issue with your local python environment or IDE.

closed 2018-12-23 by Knio

CyberJordy, 2021-04-20:
I have the same problem, did someone found the solution?