Knio/pynmea2 #84: Error on AIS messages

audiether, 2019-03-01:
My NMEA stream contains AIS messages in the following format:

!AIVDM,1,1,,A,ENkb9ONaR@@@@@@@@@@@@@@@@@@;kaD09EE>`00003vP000,4*2C

When I try to parse this message I get a ParseError.

Any way to enable parsing AIS messages?

Knio, 2019-03-05:
Hi, unfortunately we don't have support for AIS messages. If you are able to help contribute code, or even just provide references and samples, it would be appreciated!

audiether, 2019-03-08:
I doubt I'll be able to contribute much in the  way of code... I'm much more of a hack than an experienced programmer. :-)

I can definitely provide some sample messages and see what else I can find in the way of resources.  I'm on the road and need to be on the boat to get the samples, so it will be about two weeks, but I will definitely post here.

THANK YOU for pynmea2!!!

audiether, 2019-03-08:
Here are the details on the data structure:
http://catb.org/gpsd/AIVDM.html

klint-k, 2019-07-11:
I found two file on the internet. 
nmea_data_sample.txt
nmea_sample.txt
 The script that i wrote:

`with open('nmea-sample.txt','rt') as file_obj:
    nmea_obj = file_obj.readlines()

with open('nmea_data_sample.txt','rt') as file_obj:
    nmea_obj += file_obj.readlines()

bad = {}

for k in nmea_obj:
    t = k.split(',')
    #print()
    if t[-1] == '0*26\n' or t[-1] == '0*25\n':
        pass
##    elif t[0] != "!AIVDM":
##        print(t)
##        pass
    elif t[0][3:] != "VDM":
        print(t[0][3:])
        pass
    elif t[1] == '1':
        #print(t)
        try:
            x = ais.decode(t[-2], 0)
            #print(x)
        except _ais.DecodeError:
            try:
                x = ais.decode(t[-2], 1)
                #print(x)
            except _ais.DecodeError:
                try:
                    x = ais.decode(t[-2], 3)
                    #print(x)
                except Exception as e:
                    bad[k] = e
                    print(e)
                    pass
        except Exception as e:
            bad[k] = e
            print(e)
        else:
            for i in x:
                #print('%s: %s' % (i, x[i]))
                pass
    elif t[1] == '2' and t[2] == '1':
        store = t
    elif t[1] == '2' and t[2] == '2':
        try:
            #store += t
            #print(store)
            #print(t)
            x = ais.decode(store[-2] + t[-2], 2)
            #print(x)
        except Exception as e:
            bad[str(k) + str(store)] = e
            #bad.append(store)
            #bad.append(t)
            print(e)
        else:
            for i in x:
                #print('%s: %s' % (i, x[i]))
                pass
            store = ''
`

klint-k, 2019-07-11:
try this github

https://github.com/schwehr/libais

rtprio, 2019-10-06:
@Knio how would one structure the fact that ais messages can be more than one line? Keep track of the count and re-parse?  I'm not sure one can assume the next line would be the next part of the message, but I haven't seen any examples otherwise.

```
>>> foo = pynmea2.parse("!AIVDM,2,1,6,B,56:fS:D0000000000008v0<QD4r0`T4v3400000t0`D147?ps1P00000,0*3")
>>> foo.is_complete
False
>>> foo.parse("!AIVDM,2,2,6,B,000000000000008,2*29")
>>> foo.is_complete
True

```

That's just getting the whole binary blob; I'm not exactly sure how to make it magically re-split when it comes time to `.render()` it again. It seems there are assumptions than the output is a single line again.

xOneca, 2019-10-21:
(Disclaimer: I have **zero knowledge** of AIS, but maybe I don't get too far)

I would let pynmea take care of the "transport" of the binary data in chunks, as it arrives. Then I would have an "upper stage" to which feed those blobs, and let it join and interpret the data as a whole.

For rendering, I would split the binary data in chunks and then encode each chunk in its own message. But this, again, using a first stage to split, and then pynmea (if it can) to encode them.