Knio/pynmea2 #87: Runtime error on Python 3 in NMEA Stream Reader

sterwen, 2019-04-17:
<class 'bytes'> <class 'str'>
Traceback (most recent call last):
  File "GPS_nmea.py", line 112, in <module>
    main()
  File "GPS_nmea.py", line 106, in main
    pos=nmeatty.getNMEAPosition()
  File "GPS_nmea.py", line 69, in getNMEAPosition
    for msg in self._reader.next():
  File "/usr/local/lib/python3.4/dist-packages/pynmea2/stream.py", line 49, in next
    lines = (self.buffer + data).split('\n')
TypeError: Can't convert 'bytes' object to str implicitly

Knio, 2019-05-12:
`GPS_nmea.py` looks like your own code here, and not pynmea.

closed 2019-05-12 by Knio

sterwen, 2019-05-13:
Hi Tom,

I found the error in the pynmea2 code. in the file stream.py, there is one
statement that can't work in Python3 because it mix str and bytes. I have
made the correction and some simplification in the file.

I can propose that to be posted to the main.

Rgds

Laurent Carré
http://sterwen.typepad.com

sterwen, 2019-05-13:
Hello,

The bug still exist line 48 in the current version 
`lines = (self.buffer + data).split('\n')`
This will raise an error because self.buffer is type <str> while data is type <byte>
Here how I made it compatible with both versions:
`if type(data) == str :
            line = data
  else:
           line = data.decode('ascii')`
I have also removed the buffer because it is never used.

xOneca, 2019-05-13:
Actually, `self.buffer` **is** used. Specifically if `data` contains more than just one message, but last one is not yet complete.

xOneca, 2019-12-01:
I think `self.buffer` should be initialized to `b''`, instead of `''`, because it seems that `NMEAStreamReader.stream.readline()` returns type `bytes`, not `str`.