Knio/pynmea2 #102: Is this library working I cannot use the example to stream data from GPS

BetterRobotics, 2020-03-21:
I cannot work out how to use this library, I have tried everything I can think of but keep getting errors this is from the example code

> File "/home/al/.local/lib/python3.6/site-packages/pynmea2/stream.py", line 48, in next
>     lines = (self.buffer + data).split('\n')
> TypeError: must be str, not bytes

Here is the code

```
import time
import serial
import pynmea2

def read(filename):
    f = open(filename)
    reader = pynmea2.NMEAStreamReader(f)

    while 1:
        for msg in reader.next():
          print(msg)

def read_serial(filename, baud=9600):
    com = None
    reader = pynmea2.NMEAStreamReader()

    while 1:

        if com is None:
          try:
            com = serial.Serial(filename, baud)
          except serial.SerialException:
            print('could not connect to %s' % filename)
            time.sleep(5.0)
            continue

        data = com.read(16)
        for msg in reader.next(data):
          print(msg)
          
read_serial('/dev/GPS', 9600)
```

BetterRobotics, 2020-03-21:
Issue was related to python 2/3 differences I'm using python 3

I feel this is a bug though. 

The work around was to add `decode("utf-8", "ignore")` like this.. 

```
        data = com.read(16)
        data = data.decode("utf-8", "ignore")
        for msg in reader.next(data):
          print(msg)
```

Knio, 2020-03-21:
The examples were old. I have updated them in 096d2f6

closed 2020-03-21 by Knio