Knio/pynmea2 #107: Extracting data container

ptoews, 2020-04-26:
Hi,
I would like to extract the parsed values together with their field names from the NMEASentence object into a general container, for example into a namedtuple or dictionary. As far as I can see, there is no possibility to do this yet. However, the __repr__() method does pretty much exactly that, but returns a string instead. Would it be possible to separate the field/data collection and string representation, so that there is an additional method like as_dict() or similar?

ptoews, 2020-05-20:
Anything new about this? Otherwise I would be happy to help, and open a PR with the suggested approach. Or are there some special things to consider?

Knio, 2020-06-11:
This is a good idea. I haven't had much time to work on this project in a while, but I'm open to PRs!