Knio/pynmea2 #120: Wrong QuerySentence

fabien-degroote-dev, 2021-02-18:
in my opinion, the 'QuerySentence' is wrong
I have:
      $KDAIQ,VER,*17
instead of:
      $KDAIQ,VER*3B
the last ',' should be removed

class QuerySentence(NMEASentence):
    sentence_types = {}
    def __init__(self, talker, listener, sentence_type):
        self.talker = talker
        self.listener = listener
        self.sentence_type = sentence_type
        self.data = []

    def identifier(self):
        return '%s%sQ,%s,' % (self.talker, self.listener, self.sentence_type)

fabien-degroote-dev, 2021-02-18:
I have just realized that issue #119 relates the same problem

Knio, 2021-02-18:
Rationale posted on #119. Can you show an example of this being a bug?

fabien-degroote-dev, 2021-02-19:
I am not expert in NMEA even less when it comes to queries but, my ICD says that to get the version number of my device, I need to issue a '$KDAIQ,VER*3B' request.
Then, I made the following in the software: 'pynmea2.QuerySentence('KD', 'AI', req)'
The outcome is '$KDAIQ,VER,*17' as in my opinion there is an unnecessary ',' at the end of the following line of code:
           'return '%s%sQ,%s,' % (self.talker, self.listener, self.sentence_type)'

Knio, 2021-02-19:
Thanks, I understand a bit better now... this might be a bug in Query sentences, or in sentences with empty data payloads, or it could be your device manufacturer not following the spec wrt commas as many do. Are you able to link the documentation for your device? We don't have good testing around Query sentences. (Also, I am not an expert in NMEA either other than maintaining this library and I don't have any devices that support Query..)

The NMEA spec itself seems confused and shows both a trailing comma in the format but then not:
[image]

fabien-degroote-dev, 2021-02-19:
Hi, it is about an AIS transponder and the ICD tells:

> IEC standard sentences may be queried using the IEC standard query format:
> $<aa><aa>Q,<ccc>*hh<CR><LF>
> where <aa><aa> are talker identifiers of the talker and recipient respectively and <ccc>
> is the formatter identifying the requested sentence.

So, I agree it is also not clear in this manual but,
1. if I send `$KDAIQ,VER,*17` the device reports an error in the request
2. if I send `$KDAIQ,VER*3B` it works.

As far as I understand NMEA, commas are used as separator for the fields.
I suggest to 'remove' them if not necessary (i.e. when no parameter is added)

closed 2021-02-20 by Knio

Knio, 2021-02-20:
No tests failed, so I removed the comma from QuerySentence. Published as 1.16.0 (a minor version bump) because I expect this may break some devices and people may complain, but until they come forward with more test cases this seems like the best we can do.

commit 3ab85169 (2021-02-20): Remove trailing comma from QuerySentence identifier. Fixes #120