Knio/pynmea2 #122: Issues with parshing PASHR sentences

freol35241, 2021-03-26:
Hi,

Just ran into some troubles when trying to parse an NMEA stream containing PASHR sentences, like below:
```
$PASHR,122144.75,240.68,T,,,,,,,2*04
$PASHR,122144.96,240.30,T,,,,,,,2*04
$PASHR,122145.18,239.91,T,,,,,,,2*06
$PASHR,122145.61,239.04,T,,,,,,,2*04
$PASHR,122145.83,238.59,T,,,,,,,2*01
$PASHR,122146.05,238.13,T,,,,,,,2*02
```

Seems like the special logic for these messages (see [here](https://github.com/Knio/pynmea2/blob/master/pynmea2/types/proprietary/ash.py#L24-L26) and [here](https://github.com/Knio/pynmea2/blob/master/pynmea2/types/proprietary/ash.py#L35-L37)) does not fit the messages that I have. 

The regex in the the last link is not catching the messages due to different number of decimal values in the timestamp. I.e. `122144.75` does not give a match with `'^\d{6}\.\d{3}$'`. Changing the regex to `'^\d{6}\.\d{2,3}$'` seems to do the trick.

jmwooten, 2021-03-26:
Do you have a manual or receiver reference for this message?

freol35241, 2021-03-27:
Yep, here: https://anavs.com/knowledgebase/nmea-format/ (at the bottom)

jmwooten, 2021-03-30:
I looked at another manual for the [MB-Two](https://trl.trimble.com/docushare/dsweb/Get/Document-933357/MB-Two_RM_F_en_Dec%202019.pdf#page349). The ATT message there shows a similar two digit seconds in the time stamp. I think the regex has a bug. 

Feel free to submit a patch with a test.

PR #124 (2021-03-31): Fix for ashratt regex
Fixes #122
closed 2021-03-31 by Knio