Knio/pynmea2 #139: PTNL PJK File Update

arcsump, 2022-04-27:
### Discussed in https://github.com/Knio/pynmea2/discussions/138

<div type='discussions-op-text'>

<sup>Originally posted by **arcsump** April 21, 2022</sup>
Hi @Knio , 
Thanks for the awesome library, 

I needed to extend a proprietary class, wondering if it can be properly incorporated.
That would be great, many thanks again - I am not that well-versed so yeah pardon me.

Meanwhile, I modified tnl.py and added this snippet

`class TNLPJK (TNL, DatetimeFix):`
`    fields = (`
`        ('Empty', '_'),`
`        ('Sentence Type', 'type'),`
`        ('Timestamp', 'timestamp', timestamp),`
`        ("Datestamp", "datestamp", datestamp),`
`        ('Northing', 'northing'),`
`        ('Northing Direction', 'northing_dir'),`
`        ('Easting', 'easting'),`
`        ('Easting Direction', 'easting_dir'),`
`        ('GPS Quality', 'gps_quality'),`
`        ('Total number of satelites in use', 'num_sats'),`
`        ('DOP', 'dop'),`
`        ('Height of Antenna Phase Center, GHT or EHT', 'phase_center_height'),`
`  )`
    
  </div>

kamiccolo, 2024-01-26:
Mhm. Weird, I remember making a PR some time ago: https://github.com/Knio/pynmea2/pull/75

Which seems... to get merged only partially. Will see what can I do about that.

arcsump, 2024-01-26:
Hi! Thanks Appreciate it, cheers

Marcus

kamiccolo, 2024-03-07:
I believe this can be closed :) PR got merged: https://github.com/Knio/pynmea2/pull/75

Adding: `TNLDG, TNLVGK and TNLPJK`

arcsump, 2024-03-13:
Awesome! Many thanks for this!

closed 2024-03-13 by arcsump