Knio/pynmea2 #155: Micropython compatability

rjenningsuk, 2023-02-16:
Is it possible to make this compatible with micro python?

Currently on installing it and the first run I'm straight away hit with:

Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
  File "/lib/pynmea2/__init__.py", line 8, in <module>
  File "/lib/pynmea2/nmea.py", line 29, in <module>
  File "/lib/pynmea2/nmea.py", line 20, in __init__
AttributeError: type object 'type' has no attribute '__init__'

QiaoTuCodes, 2023-09-09:
Yeah! it's not use for micropython . that's a big lib on x86&64 platform . i mean you can write small codes on micropython or , talk a look this share: https://github.com/coylen/upyNMEA

kamiccolo, 2024-01-26:
> Is it possible to make this compatible with micro python?
> 
> Currently on installing it and the first run I'm straight away hit with:
> 
> Traceback (most recent call last): File "", line 2, in File "/lib/pynmea2/**init**.py", line 8, in File "/lib/pynmea2/nmea.py", line 29, in File "/lib/pynmea2/nmea.py", line 20, in **init** AttributeError: type object 'type' has no attribute '**init**'

Could You please write down a brief instruction how to reproduce it, for people who may not be very proficient with MicroPython and it's infrastructure?