Knio/pynmea2 #162: RMC message error in 1.19.0 (but 1.15.0 works fine)

comperem, 2023-12-30:
Hello,

There seems to be a problem with v1.19.0 `RMC` messages.

pynmea 1.15.0 works fine with this RMC string:

```
data='$GNRMC,200728.50,A,2829.34783,N,08122.21257,W,0.057,,301223,,,A,V*0F'
msg=pynmea2.parse(data)
```
`msg.is_valid` returns `True` with v1.15.0.

So far, so good. This is expected behavior.

But pynmea 1.19.0 says this RMC string is invalid.

`msg.is_valid` returns `False`. ???

This data is fine. I'll stick to v1.15.0 until 1.19.0 is fixed.

---
fwiw, this is the parsed data structure using 1.15.0:
```
<RMC(timestamp=datetime.time(20, 7, 28, 500000), status='A', lat='2829.34783', lat_dir='N', lon='08122.21257', lon_dir='W', spd_over_grnd=0.057, true_course=None, datestamp=datetime.date(2023, 12, 30), mag_variation='', mag_var_dir='') data=['A', 'V']>
```

this is the parsed data structure using 1.19.0:
```
<RMC(timestamp=datetime.time(20, 7, 28, 500000, tzinfo=datetime.timezone.utc), status='A', lat='2829.34783', lat_dir='N', lon='08122.21257', lon_dir='W', spd_over_grnd=0.057, true_course=None, datestamp=datetime.date(2023, 12, 30), mag_variation='', mag_var_dir='', mode_indicator='A', nav_status='V')>
```

Is this a bug? What's the going on here? The data is fine.

MDC

kamiccolo, 2024-01-26:
Weird, I can't see `1.19.0` here (upstream), only `1.18.0`: https://github.com/Knio/pynmea2/releases
But on PyPi it's already `1.19.0`: https://pypi.org/project/pynmea2/#history

@Knio You're the maintainer of PyPi package, right? Would You care to comment?

kamiccolo, 2024-01-26:
Agh, it seems that only version in `setup.py` was updated. Last additions to `RMC` parsing:
https://github.com/Knio/pynmea2/commit/988c297ce82d976db9094b435a1aa290e7d5b9ed

kamiccolo, 2024-01-26:
I'd say this ticket can be closed.

closed 2024-03-02 by Knio

comperem, 2024-03-04:
This is absolutely a bug if the data is good but version 1.19.0 returns invalid status.

How can you close an issue when the results are incorrect?

@Knio, please re-open this ticket and address the bug.

that's the whole point of opening an issue

kamiccolo, 2024-03-04:
> This is absolutely a bug if the data is good but version 1.19.0 returns invalid status.

It's not about "data being invalid". But rather about Navigational Status being invalid, as reported by the GNSS receiver. I believe that for validation of the the data itself, You should pass `check=True`: https://github.com/Knio/pynmea2/blob/master/pynmea2/nmea.py#L90C32-L90C34 and catch the corresponding exceptions.