Knio/pynmea2 #171: Furuno messages incomplete

Shal-Ziar, 2024-11-19:
@JoepdeJong 
I assume your changes have made it into version 1.19.0?
If so it seems that your addition of the furuno messages isn't all encompassing. The fields don't seem to be fully defined. Based on my webcrawling the values and limits  are based on:
[manual](https://www.manualslib.com/manual/1157050/Japan-Radio-Co-Jlr-21.html?page=132#manual)

Attitude message:
 ATT (Proprietary): incl. yaw, pitch roll data [0, 359.9], [-90, 90], [-90, 90]
 $PFEC,GPatt,xxx.x,+xx.x,+xx.x*hh<CR><LF> (Ver. 2.0)
sample: "$PFEC,GPatt,194.483,-0.963,17.172*58"

Heave message:
    HVE (Proprietary): incl. heave data unit in m [-99.999,  99.999], A/V
    $PFEC,GPhve,xxx.x,A*hh<CR><LF> (Ver. 2.1)
    "$PFEC,GPhve,-0.114,A*16",

Additionally I'm still looking for a description of the IMU message GPimu
"$PFEC,GPimu,0.210,0.056,-0.023,0,-0.654,0.448,0.486,0,0*42",

I'm planning on 'reverse engineering' GPimu, but that will require some vessel data analysis and might take some time.

JoepdeJong, 2024-11-20:
It appears there may be differences in the versioning of NMEA messages. I am not certain how to tackle that..

Also, I have been unable to find documentation for the `$PFEC,GPimu` NMEA sentence for Furuno devices. However, I have come across implementations on GitHub that utilize the standard `$GPimu` sentence. https://github.com/search?q=%22%24GPimu%22&type=code

Based on these implementations, the `GPimu` sentence is formatted as follows:

$GPIMU | time of week | accel-x | accel-y | accel-z | gyro-x | gyro-y | gyro-z
-- | -- | -- | -- | -- | -- | -- | --

This might help with reverse engineering the `GPimu` field.

Shal-Ziar, 2024-11-20:
Thanks, 
I indeed expect that the gyro and accelerator components are in the message. However, based on my data-collection it doesn't seem to contain time_of_week:
```
    "$PFEC,GPimu,0.210,0.056,-0.023,0,-0.654,0.448,0.486,0,0*42",
    "$PFEC,GPimu,0.185,0.028,-0.012,0,-0.707,0.338,0.619,0,0*45",
    "$PFEC,GPimu,0.184,0.023,-0.012,0,-0.257,0.246,0.768,0,0*40",
    "$PFEC,GPimu,0.181,0.052,-0.036,0,-0.150,0.551,0.993,0,0*4A",
    "$PFEC,GPimu,0.197,-0.007,-0.006,0,-0.242,0.345,1.016,0,0*65",
    "$PFEC,GPimu,0.204,0.013,-0.035,0,-0.112,0.540,1.054,0,0*47",
    "$PFEC,GPimu,0.207,-0.005,-0.081,0,-0.028,0.524,1.104,0,0*6F",
    "$PFEC,GPimu,0.254,0.028,-0.069,0,-0.009,0.593,1.111,0,0*46",
    "$PFEC,GPimu,0.214,-0.003,-0.023,0,-0.024,0.692,1.165,0,0*66",
    "$PFEC,GPimu,0.208,0.010,-0.093,0,-0.161,0.673,1.180,0,0*4B",
    ```

I'll see if I can reach-out to a contact that may have this information. If so I'll see If I can make a pull request.